KiCad/kicad-i18n #2: Missing compilation of .mo files from .po source files

ciampix, 2015-09-20:
In actual cmake config files is missing the automatic generation of compiled gettext message catalogs (.mo) files from their source (.po) files.

LordBlick, 2015-09-20:
Can you provide some build log with fail? I've build always from clean tree and I've no problems occurred. My changes not touch *.po→*.mo transformations.

ciampix, 2015-09-20:
I do:
cmake .
-- Configuring done
-- Generating done
-- Build files have been written to: /home/marco/git/github/kicad-i18n
marco@ciampix-A532:~/git/github/kicad-i18n$ make
marco@ciampix-A532:~/git/github/kicad-i18n$ rm it/kicad.mo
marco@ciampix-A532:~/git/github/kicad-i18n$ make
marco@ciampix-A532:~/git/github/kicad-i18n$ sudo make install
[sudo] password for marco: 
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/share/kicad/internat/bg/kicad.mo
-- Installing: /usr/local/share/kicad/internat/ca/kicad.mo
-- Installing: /usr/local/share/kicad/internat/cs/kicad.mo
-- Installing: /usr/local/share/kicad/internat/de/kicad.mo
-- Installing: /usr/local/share/kicad/internat/el/kicad.mo
-- Installing: /usr/local/share/kicad/internat/es/kicad.mo
-- Installing: /usr/local/share/kicad/internat/fi/kicad.mo
-- Installing: /usr/local/share/kicad/internat/fr/kicad.mo
-- Installing: /usr/local/share/kicad/internat/hu/kicad.mo
CMake Error at cmake_install.cmake:72 (FILE):
  file INSTALL cannot find "/home/marco/git/github/kicad-i18n/it/kicad.mo".

make: *** [install] Errore 1

what is wrong?

LordBlick, 2015-09-20:
Its bad idea to use same dirs with input and output. Usually it is recommended creating a separate folder for the output files.  `` cmake``` creates a lot of extra files that hinder subsequent amendments. Removal of all external folder is much simpler.
Try:
```
[me ~/]$ mkdir -p ~/tmp/kicad-i18n-build
[me ~/]$ cd ~/tmp/kicad-i18n-build
[me ~/tmp/kicad-i18n-build]$ cmake ~/git/github/kicad-i18n
-- Found Gettext: /usr/bin/msgmerge (found version "0.19.5.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: ~/tmp/kicad-i18n-build
[me ~/tmp/kicad-i18n-build]$ DESTDIR=~/tmp/kicad-i18n make install
Install the project...
-- Install configuration: ""
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/bg/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/ca/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/cs/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/de/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/el/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/es/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/fi/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/fr/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/hu/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/it/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/ja/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/ko/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/nl/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/pl/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/pt/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/ru/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/sk/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/sl/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/sv/kicad.mo
-- Installing: ~/tmp/kicad-i18n/usr/local/share/kicad/internat/zh_CN/kicad.mo
```
I've never install anything on my system without package manager(you can throw yourself inadvertently OS on the back), so I've use ```DESTDIR``` environment variable to bypass the need to use sudo.

ciampix, 2015-09-20:
OK done this (BTW many thanks for your patience...): 
marco@ciampix-A532:~$ mkdir -p ~/tmp/kicad-i18n-build
marco@ciampix-A532:~$ cd ~/tmp/kicad-i18n-build/
marco@ciampix-A532:~/tmp/kicad-i18n-build$ cmake ~/git/github/kicad-i18n/
-- Configuring done
-- Generating done
-- Build files have been written to: /home/marco/git/github/kicad-i18n
marco@ciampix-A532:~/tmp/kicad-i18n-build$ make
make: *** No targets specified and no makefile found.  Stop.
... I have a doubt ... let me see ...
marco@ciampix-A532:~/tmp/kicad-i18n-build$ rm -r ~/git/github/kicad-i18n/CMakeFiles
marco@ciampix-A532:~/tmp/kicad-i18n-build$ rm ~/git/github/kicad-i18n/CMakeCache.txt
marco@ciampix-A532:~/tmp/kicad-i18n-build$ rm ~/git/github/kicad-i18n/cmake_install.cmake
marco@ciampix-A532:~/tmp/kicad-i18n-build$ rm ~/git/github/kicad-i18n/install_manifest.txt
marco@ciampix-A532:~/tmp/kicad-i18n-build$ cmake ~/git/github/kicad-i18n/
-- Found Gettext: /usr/local/bin/msgmerge (found version "0.19.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/marco/tmp/kicad-i18n-build
marco@ciampix-A532:~/tmp/kicad-i18n-build$ make
marco@ciampix-A532:~/tmp/kicad-i18n-build$ DESTDIR=~/tmp/kicad-i18n-root make install
Install the project...
-- Install configuration: "Release"
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/bg/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/ca/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/cs/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/de/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/el/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/es/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/fi/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/fr/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/hu/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/it/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/ja/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/ko/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/nl/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/pl/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/pt/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/ru/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/sk/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/sl/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/sv/kicad.mo
-- Installing: /home/marco/tmp/kicad-i18n-root/usr/local/share/kicad/internat/zh_CN/kicad.mo
ok

Issues:

1) there is no a "make clean" or "make distclean" target the erase all build files, Makefiles and cmake cache/build files
2) make does nothing: the actual build is done when I type
cmake ~/git/github/kicad-i18n/
and actually cmake told me so:
[...]
-- Build files have been written to: /home/marco/tmp/kicad-i18n-build
##^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
the "make" command does nothing at all! 
... it is weird to me ... perhaps cmake config files are wrong or something else?

LordBlick, 2015-09-20:
Ad. 1 It can't be done by make, because make files needs to be removed too. http://stackoverflow.com/questions/9680420/looking-for-a-cmake-clean-command-to-clear-up-cmake-output Feel free to implement it, if you need, but building in external directory and removing it after install seems to be enough.
Ad.2 If you over think about it, there is nothing to do with a `make`(Look into `~/tmp/kicad-i18n-build/Makefile` and `~/tmp/kicad-i18n-build/CMakeFiles/Makefile2`), because all we need is in `CMakeLists.txt`. We do not compile and link anything in addition, only install.
At least it works… ;)

ciampix, 2015-09-21:
First of all, thanks.
1. I agree, it seems the best solution.
2. I agree too but it seemed to me strange at first because:
  a) I am not used to use cmake and I tried (without results) to re-create .mo files with make
  b) I was following the README that reports the use of make

ciampix, 2015-09-21:
PS: I think that this ER could be closed.

LordBlick, 2015-09-21:
After some thought, I open a bug report again, it would lead to the building output files by `make`. The key here seems to be command `add_custom_command`, instead of `execute_process`.
That also needs to add some `make` targets using this.

ciampix, 2015-12-02:
@LordBlick are you working on this or I can close it for now?

ciampix, 2018-01-01:
Reopened in #208 with another title... closing it.

closed 2018-01-01 by ciampix