KiCad/kicad-i18n #3: Translations license

rdmitry, 2015-09-24:
There is no license file for translations. Is it GPL-2.0+ as KiCad or KiCad documentation license?

ciampix, 2015-09-24:
Oh my ... you are right! We have to contact all translators to add this string to the po header as I have done with the kicad manuals translations:

# Copyright (C) YEAR Free Software Foundation, Inc.
# This file is distributed under the same license as the KiCad package.

LordBlick, 2015-09-24:
That's not FSF Copyright, but only licence…

ciampix, 2015-09-24:
Ok but a disclaimer should be present anyway...

LordBlick, 2015-09-24:
```
Copyright (C) 2004-2015 KiCad Translation Team, see contributors.txt for contributors.
# This file is distributed under the same license as the KiCad package.
```
We need to add contributors.txt

ciampix, 2015-09-24:
I do not think we should. Usually .po files has this kind of rows just below (FSF deleted):

 # Cvpcb User Manual Italian Translation
 # Copyright (C) 2014-2105 KiCad Translation Team.
 # This file is distributed under the same license as the KiCad package.
 # Marco Ciampa <ciampix@libero.it>, 2014-2015.
 # [Other translator...]
 # ...

so any translator add himself to the list...

LordBlick, 2015-09-24:
Translators info has proprietary place in .po file header, there's no need to repeat.
```
"Last-Translator: LordBlick <LordBlick by gmail>\n"
"Language-Team: Polish KiCAD Team: Mateusz Skowroński, Krzysztof Kawa, Kerusey Karyu, Daniel Dawid Majewski <LordBlick>\n"
```

LordBlick, 2015-09-24:
Documentations usually has differ licence:
http://www.gnu.org/doc/doc.en.html

LordBlick, 2015-09-26:
If there's no objections, I will close this issue in several hours.

ciampix, 2015-09-26:
IMHO the "language team" is not sufficient, if not wrong sometimes. Better option is to explicitly list the translators at the end of the header comments.
To refer to a documentation license is a good idea though.

LordBlick, 2015-09-27:
IMHO better will be make separate file than pasting not necessary comments. We do not have guarantee, that some version of poedit can cut those down…

ciampix, 2015-09-28:
Actually poedit use gettext (it uses msgfmt internally) and this is a guarantee that it do not and will not change unnecessarily the .po headers and comments. Anyway I am trying to get rid of poedit .po compilation function altogether to enable the use of other .po editors (emacs, vi, gtranslator, gted, lokatize, virtaal, et al.) so the translator copyright could stay in the .po file safely without the need to add another file.

LordBlick, 2015-09-28:
IMHO it still have no sense. Comments are not accessible in poedit, but „Language-Team” is. „Language-Team” is in file header, so if we view file in any editor, is good enough visible and easy to find. Any other information is unnecessary cluttering of the file. If we really want to honor the translators of earlier versions, easier and better to include their names in a specially created file.

slavkoja, 2016-02-01:
IMO, if there will be POT file generated by upstream, these comments can be added into it and will be added by next PO file update from POT template.

ciampix, 2016-09-05:
Sorry for a so late reply. 
Usually updated strings, even comments, should not be modified by the template POT file. POT file are just these: templates. A template main usefulness is the first time you start a translation. Then the po files are updated and POT file are referred only for time-to-time po files updates for new strings. Old strings on .po files are never touched but fuzzied if not found a corresponding match on the on-the-fly generated pot file for .po file update only use. Comments should be _never_ touched by po files update procedures to avoid deleting useful translation notes putted on .po files by translators. So putting comments into POT files are useful just for the first time .po file generation and then no more.

ciampix, 2018-07-28:
This is way too old, I think all relevant issues are answered or not so important. May we close this now?

closed 2019-01-20 by ciampix