KiCad/kicad-i18n #6: Translation file is too large!

yoneken, 2015-09-28:
In the kicad.po file, there are approximately 4.8k items to be translated.
Sometimes, gete keepers of the repository have to check a vast number of them.
It's almost impossible.

How about to consider to separate the translation file to each application? (e.g., eeschema.po, pcbnew.po, ...)

ciampix, 2015-09-28:
I do not understand your problem here. If you use a decent .po editor, it provides you just the untranslated strings that are really very few from one commit to another... 

Have you tried to use emacs as a .po file editor (in po-mode)?
I find poedit rather unusable with large .po files and not a problem with emacs...I now use poedit just to merge/update the strings until we find a way to do it automatically with the cmake toolchain...

LordBlick, 2015-09-28:
Menu „Catalog” → „Purge deleted translations”

yoneken, 2015-09-28:
Poedit sometimes changes the order of reading file while updating process.
It makes a big patch. (e.g. https://github.com/KiCad/kicad-i18n/commit/addc28012190c299d2a4a0792db685d8ea111435 )
We confirmed the issue is reproduced with Poedit 1.7.x in Ubuntu and Mac OSX .
It's not a bug of KiCad, but it affects our project.

Actually, I don't use Poedit.
However, Poedit is a convinient tool to commit translation for non-programmers...

ciampix, 2015-09-29:
Me neither use poedit.
That's another reason to get rid of poedit update function altogether and use cmake+gettext instead.

jp-charras, 2015-09-29:
Looks like some Japanese translations are made with Poedit 1.8.5 and
some other with 1.7.5

Perhaps this is the reason for these large diff files.

I am using Poedit and I have not noticed large diff files.

What do you call "poedit update function" ?
AFAIK, Poedit uses gettext to extract strings.

ciampix, 2015-09-29:
poedit use gettext but the order of reading the files to feed the msgfmt executable is what created this mess. If we compile the executable list and using cmake use gettext to compile the .mo files (we already do so) and update the .po file (I am working on it) we can get rid of this gettext "button". In this way the order of the list will not change any more and we can use the po editor of choice, even a web version...

yoneken, 2015-10-05:
Thank you for the discussion.
I will re-open the issue if the number of the words to be translate were much more.

closed 2015-10-05 by yoneken