KiCad/kicad-i18n #45: Misiing Slovak (sk) in Language menu

slavkoja, 2016-02-01:
I don't know if this is proper place, if not then i am sorry. But recently the Slovak translation was added, however the Slovak language is missing in the Language menu.

The Slovak translation works for me when i choose the default language, but i afraid what when the system's language differs.

ciampix, 2016-02-10:
I think it is better to open a bug report on Launchpad and signal this on dev list too. I have not write permission on the main KiCad source repo.

slavkoja, 2016-02-10:
I understand this. But, i have no account on launchpad and i cannot create the Ubuntu one account (because there is only "I am new Ubuntu user" only and i am not new nor Ubuntu user ;-) ).

Please, can you report it? But, be free to close this...

jp-charras, 2016-02-12:
Le 10/02/2016 17:49, Slavko a écrit :
> I understand this. But, i have no account on launchpad and i cannot
> create the Ubuntu one account (because there is only "I am new Ubuntu
> user" only and i am not new nor Ubuntu user ;-) ).
> 
> Please, can you report it? But, be free to close this...
> 

I added the Slovak entry in language menu in rev 6550.
Please, test it.

Language entries are no needed to use translations (setting language to
default is enough), but they are very useful for tests and to know what
translations are available.

-- 
Jean-Pierre CHARRAS

slavkoja, 2016-02-12:
Thanks a lot!

You are right, in standard situation, the language menu is not needed, but i know a lot of Slovak people, which are using different system locales for different reasons and they cannot switch KiCAD to Slovak then...

closed 2016-02-12 by eldarkg