KiCad/kicad-i18n #103: The status tool give the wrong status (ru, branch 4.0)

eldarkg, 2016-08-03:
I use to translate the Poedit tool.
It show that all strings translated and no fuzzy.
But if I run the status tool it show me:
## ru
4894 translated messages
12 fuzzy translations
14 untranslated messages

eldarkg, 2016-08-03:
Poedit 1.8.7.1, OS Ubuntu 16.04 LTS

nickoe, 2016-08-03:
@eldarkg yeah, but if you open poedit on it after the tool has been run it will show the same. Can you share your poedit settings?

nickoe, 2016-08-03:
If I look in the `ru/kicad.po` I see:
```
$ grep -r "Poedit-SearchPath" ru/
ru/kicad.po:"X-Poedit-SearchPath-0: 3d-viewer\n"
ru/kicad.po:"X-Poedit-SearchPath-1: bitmap2component\n"
ru/kicad.po:"X-Poedit-SearchPath-2: common\n"
ru/kicad.po:"X-Poedit-SearchPath-3: cvpcb\n"
ru/kicad.po:"X-Poedit-SearchPath-4: eeschema\n"
ru/kicad.po:"X-Poedit-SearchPath-5: gerbview\n"
ru/kicad.po:"X-Poedit-SearchPath-6: kicad\n"
ru/kicad.po:"X-Poedit-SearchPath-7: pagelayout_editor\n"
ru/kicad.po:"X-Poedit-SearchPath-8: pcb_calculator\n"
ru/kicad.po:"X-Poedit-SearchPath-9: pcbnew\n
```
and if I compare that with the POTDIRS file in the repo I see:
```
$ cat POTDIRS 
kicad
pcbnew
common
eeschema
cvpcb
3d-viewer
gerbview
bitmap2component
new
polygon
pcb_calculator
potrace
tools
pagelayout_editor
include
lib_dxf
```
I guess that is the reason for the mismatch?

eldarkg, 2016-08-03:
Fixed in 23e033a

closed 2016-08-03 by eldarkg