KiCad/kicad-i18n #116: 4.x as master

slavkoja, 2016-11-06:
Please, make the 4.0 branch as master. IMO, it is confusing, when the default (master) branch doesn't show current status and one need to switch the branch after clone. Or i miss something?

ciampix, 2016-11-06:
IMHO your suggestion is a bad idea. Master is the dev branch and the i18n repository is organized likewise the git repository of source code. Why the i18n repository should differ from the source code repository it correspond?

slavkoja, 2016-11-07:
OK, i was only surprised, when i take look on sk translation status and on the default page it is 2 years old, but there was new added. Then i find it under the 4.0 branch. Then my main idea was, that the default branch have provide current status, but i never take look into the kicad's source repo organization. It makes sense now. I am sorry for noise ;-)

closed 2016-11-07 by slavkoja