KiCad/kicad-i18n #207: Problem opening project files (at least with -de template)

Treczoks, 2018-01-01:
It sounds strange, but when I use the -de template for German, opening a project does not work. When I switch back to English, it works.

This is the bug as originally reported in the KiCAD bug tracker with all the information. After getting the information that this seems to be locale related I was directed here to report it. I have personally verified that it actually is an i18n issue by trying to open a project both with the German as well as the English setting (Default is German, I had to explicitly set it to English to make it work for me).

https://bugs.launchpad.net/kicad/+bug/1740481

ciampix, 2018-01-01:
I should have fixed it in master branch...

nickoe, 2018-01-03:
@Treczoks please verify

Treczoks, 2018-01-03:
Thanks for the reminder. I updated the nightly Ubuntu packages, and behold, it works! Thanks!

closed 2018-01-03 by Treczoks