KiCad/kicad-i18n #208: I18n building should follow standard conventions

ciampix, 2018-01-01:
After this comment: https://github.com/KiCad/kicad-i18n/issues/2#issuecomment-142081524
I decided to re-open this issue with another title.
I18n building process does not follow the standard rules:
cmake .. (should configure but actually it compiles)
make (should compile but actually does nothing)
make install (actually installs, this is correct)

LordBlick, 2018-01-19:
What do you expect from configuration? Another level of inception? Is it really worth unless cmake is able to do same in just one easy step?

ciampix, 2018-01-20:
I know that it seems to be pointless but I would like to see make to compile in place and make install to install... is it asking to much?

ciampix, 2018-05-24:
If you want another reason why this should be fixed... then think about this: now you have to make a "make install" to compile and test for errors during compilation of .po files... compilation/test should always be a separate stage from install ...

StefanBruens, 2018-05-27:
Should not be to hard to do it properly ... Can someone write down what exactly is expected here? Just "mkdir build; cd build; cmake .. ; make; make install"? Any configuration options, additional make targets?

ciampix, 2018-05-28:
"mkdir build; cd build; cmake .. ; make; make install" should suffice IMHO