KiCad/kicad-i18n #231: Italian translation fails to compile

GyrosGeier, 2018-03-16:
Since commit eabd17e67ac048490ba8c9264d4894b33fa4ffe3, msgfmt fails while processing the Italian translation.

nickoe, 2018-03-16:
I can confirm this on the fedora builds:
```
-- Found Gettext: /usr/bin/msgmerge (found version "0.19.8.1") 
/builddir/build/BUILD/kicad-r12433-fb31bc1c/kicad-i18n-r12433-fb31bc1c/it/kicad.po:12188: duplicate message definition...
/builddir/build/BUILD/kicad-r12433-fb31bc1c/kicad-i18n-r12433-fb31bc1c/it/kicad.po:12115: ...this is the location of the first definition
/usr/bin/msgfmt: found 1 fatal error
```

nickoe, 2018-03-16:
@ciampix Are you able to take a look at this?

ciampix, 2018-03-16:
Sorry fixed right now.

closed 2018-03-16 by ciampix

nickoe, 2018-03-16:
Aha, ok. Thank you.