KiCad/kicad-i18n #280: Commit message format

baranovskiykonstantin, 2018-06-15:
Do we need a common format of commit message?
And what information may be helpful in this message? 
For now every translator write messages as they wont and and to check the state of translations not so easy.
Maybe we should arrange to use common style of commit messages?
For example:
update from last sources:
[RU] update for b37969f584c900eda07748fb198daab8a261930a
fix some errors in translation without updating from sources:
[RU] fixes (... optional explanation)
or something else.

ciampix, 2018-06-15:
On Thu, Jun 14, 2018 at 11:12:31PM -0700, Konstantin wrote:
> Do we need a common format of commit message?
> And what information may be helpful in this message?
> For now every translator write messages as they wont and and to check the state of translations [...]

Yeah, we discussed this a some time ago:

https://lists.launchpad.net/kicad-developers/msg27082.html

and the reply that we can follow of course:

https://lists.launchpad.net/kicad-developers/msg27086.html

-- 

Marco Ciampa

I know a joke about UDP, but you might not get it.

------------------------

 GNU/Linux User #78271
 FSFE fellow #364

------------------------

ciampix, 2018-06-15:
To be honest, since these logs are limited to a very restricted area, I do not see any problem in such generic messages.
The only issue I see is only related to the state of the translations and the update of the message strings. From time to time it could be useful to update all languages strings just to have an updated outlook on the status of the translations. I never did it just to avoid conflicts but we might agree on a regular scheduled update, for example the first week of the month. What do you think about this?

baranovskiykonstantin, 2018-06-15:
I think that our main problem that all translations synced with different commits of sources.
We should to use common origin, for example to use tags.
I mean that after the sources was tagged (for now it 5.0.0-rc2) all translations have to be synced with this tag (this work must be done by maintainer or an automatic script, but not by translator) and all translators shold work with untranslated strings (or fixing somethings).
When the sources will be tagged as 5.0.0 (release) all translations must be synced again and translators must update it.
Off course, after source freezing (tagging) we have to provide some time for translating.
So we will can to track state of translations by count of untranslated strings.
It is just IMHO.

ciampix, 2018-06-15:
Good point! Next time we should update the strings after the string freeze (after a warning)...

baranovskiykonstantin, 2018-06-18:
For sum up.
There are no special requirements for commit messages of translations.
But we should to follow common recommendations:
https://git.launchpad.net/kicad/tree/Documentation/development/commit-message-format.md

closed 2018-06-18 by baranovskiykonstantin