KiCad/kicad-i18n #356: "Choose the library table to add the library"

kenneth558, 2019-02-22:
As you can see, the final word "to" has been left out.  For newbies to KiCad like me, I found this yet another unnecessary confusing step on my way to learning KiCad.  Thank you for correcting this!

ciampix, 2019-02-22:
Please tell us something more... which manual? Which version? Why not create a pull request?

kenneth558, 2019-02-22:
(At my GitHub experience level, I figure if I would create a pull request, whatever that entails and it would be my very first and without hardly ANY known space for compiling on my system, I seriously doubt I would get it right.)  To find the code line culprit, all I do is a Google search on that phrase and I see a couple hits occurring without the word "to" on the end.  The 7th Google hit leads me to Line 9076 this repository page:

https://github.com/KiCad/kicad-i18n/blob/master/ca/kicad.po

The 8th and 9th hits are similar:
https://raw.githubusercontent.com/KiCad/kicad-i18n/master/ca/kicad.po
https://github.com/KiCad/kicad-i18n/blob/master/sl/kicad.po

The last hit is Slovenian language page, maybe pulled from yours or vice-versa?

Manual and Version info you request from me seems to be shown in some way at the beginning of that file for you very easily to determine, but are not listed explicitly enough for me to know. 

I am USA, so I honestly suspect the KiCad that I installed (5.0.2-bee76a0~70~Ubunto16.04.1 release build, judging from the "About" info) is of yet another source that a simple Google search didn't even show me, so  I'm just pointing out that your code has the same error as I got myself into.  How widespread this is remains to be discovered.

ciampix, 2019-02-23:
Ah sorry my bad, since I receive the issues for manuals and i18n strings, I thought it was referred to a manual, not a program string. Ok I checked the sources and it is already fixed in the upcoming version 5.2 and so will also be fixed in the next version 6. Thanks for the heads up, I'm closing this issue.

closed 2019-02-23 by ciampix