KiCad/kicad-i18n #378: Tag 5.1.1

nickoe, 2019-04-15:
I intent to dag 5.1.1 tonight in my timezone.

@ciampix Should we run the `update-po-files.sh` script first to make sure it matches the source code?

ciampix, 2019-04-15:
There is no need to run it now since strings that do not match during run-time are not loaded and that's all (no errors rising). That operation should ideally be done every now and then and just after a string freeze to let translators know that there are some untranslated/fuzzy strings to update. You (or I) can run it but now I am afraid that it is almost useless because of the tag that "freeze" the untranslated strings without any hope to incorporate the translated strings into the packages... am I wrong?

nickoe, 2019-04-15:
It looks like we got some more translations merged, I think we can tag very soon.

The following languages are complete or almost complete:
de, en, fr, it, pt, ru, pl, zh_CN, ja, and cs

Contributions that matches up with the 5.1.1 tag of the kicad source go on the 5.1 branch of kicad-i18n.

nickoe, 2019-04-15:
@MichalJahelka as you contributed last time for **cs** language I would like to poke you. Do you intend for want any updates for kicad 5.1.1, there are:

- 5455 translated messages
- 101 fuzzy translations
- 50 untranslated messages

nickoe, 2019-04-15:
@starfort-jp as you contributed last time for the **ja** language I would like to poke you. Do you intend or want any updates for kicad 5.1.1, there are:

- 5573 translated messages
- 26 fuzzy translations
- 7 untranslated messages

nickoe, 2019-04-15:
@kubasaw and @keruseykaryu as you contributed last time for the **pl** language I would like to poke you. Do you intend or want any updates for kicad 5.1.1, there are:

- 5573 translated messages
- 26 fuzzy translations
- 7 untranslated messages

nickoe, 2019-04-15:
@taotieren and @CNLiuGuang as you contributed last time for the **zh_CN** language I would like to poke you. Do you intend or want any updates for kicad 5.1.1, there are:

- 5564 translated messages
- 28 fuzzy translations
- 14 untranslated messages

nickoe, 2019-04-15:
Remember to commit on the 5.1 branch of kicad-i18n and update against the 5.1.1 tag of the kicad source.

ciampix, 2019-04-15:
Sorry if I was perhaps not clear. I thought you were about to tag (and so freeze) the translation repo. If there is time then the string update has sense.

nickoe, 2019-04-15:
I will give the translators a chance to catch up.

I does not make much sense to tag the translations on the same day as the kicad source.

stambaughw, 2019-04-15:
@nickoe I froze the strings several weeks ago so we could get the translations updated while we were squeezing in those last few bug fixed that didn't require any string translations.  If we violated that, I apologize for that.  If this isn't an acceptable solution, I will go back to the previous method and tag the source code and then give the translators a few weeks to finish up.

ciampix, 2019-04-15:
No @stambaughw you devs did not violate anything. It was probably our fault. At the announcement of the string freeze we should have updated the .po files because it is not automatic and it make sense once the strings are frozen to let translators update their translations. Hope next time we will not forget.

keruseykaryu, 2019-04-16:
I think I'm done (it properly). https://github.com/KiCad/kicad-i18n/pull/386

The branches, tags, switching between them back and forth it's too much for my head. Sorry.

nickoe, 2019-04-16:
@keruseykaryu yes, it looks correct. Thanks.

nickoe, 2019-04-16:
@stambaughw and @ciampix Please continue as is or start using a rc tags for every release. As a last minute thing I noticed that there was some translations there were fuzzy and untranslated from languages that were complete for 5.1.0. I though that we had a string freeze on the 5.1 branch since the 5.1.1 tag...

But I admit I have been sort of busy lately, so I may have missed something on the list. So up until yesterday I actually thought there were no updates required for complete translations.

ciampix, 2019-04-16:
To all contributors. Just for the sake of clarity. We all have a life, and we do this for free in the spare time. I, for a start, make many mistakes and I too have missed many, many things that many other persons have filled in (@nickoe most of the time, thanks!), and compensated, with their contributions. So when I say that there is something wrong, it is never a critic, nor pointing someone for their fault. It is just to help improve the results of our effort. Hope this is clear.

taotieren, 2019-04-16:
> @taotieren and @CNLiuGuang as you contributed last time for the **zh_CN** language I would like to poke you. Do you intend or want any updates for kicad 5.1.1, there are:
> 
> * 5564 translated messages
> * 28 fuzzy translations
> * 14 untranslated messages

OK.Thanks

starfort-jp, 2019-04-16:
@nickoe , thank you for the notice.  I'm sorry.  I think it's OK. (#389)

nickoe, 2019-04-16:
Tag pushed.

closed 2019-04-16 by nickoe