KiCad/kicad-i18n #392: Persisting and recurring quality issues in the German translation

roybaer, 2019-04-17:
What can we do about the extremely heterogeneous quality of German translated messages?

While roughly half of them are flawless, many are ungrammatical or use misspelled compound nouns.
(German compound nouns are _never_ spelled with whitespace!)

Furthermore, some lexically correct translations are semantically closer to a riddle than to a translation. (I.e. "Try to guess the English phrase")
One example is "Zoomauswahl". It is a valid translation of "zoom selection", but that function will not let you select a zoom level, it will let you zoom to a selection.

Mixing English and German acronyms for the same thing does not help, either.
(In this case "VIA" and "DuKo")

The menu entries should also either all use infinitive or they should all use imperative, but not a wild mixture.
They mostly look the same in English. (e.g. "[to] edit the file" vs. "[Computer?] Edit the file[!]")
In German, however, they are different and infinitive is way more common in German GUIs.

All this leads me to the question:
How can we address this and how can we prevent these issues in the future?
Maybe one could put a glossary and translation guidelines in a document somewhere.
Let's discuss!

nickoe, 2019-04-17:
I hope @tijuca and @GyrosGeier would like to reply to this.

GyrosGeier, 2019-04-17:
Agreed, I've also noticed that but didn't find the time to make new ones. We definitely need a compendium.

GyrosGeier, 2019-04-17:
Okay, writing a compendium is definitely hard.

I've started to collect some ideas, the main problem we have is that correct German translations are often quite long, which makes changes layout requirements and is still difficult to read because the information density isn't that great.

 - Board → Leiterplatte / Platine
 - Trace → Leiterbahn
 - Via → Durchkontaktierung

We could possibly agree on leaving "Via" untranslated, and go with the shorter "Platine" for "Board", but I'm neither a PCB design nor a UX expert.

tijuca, 2019-04-18:
Translating, or better keeping the translations up2date in the modules of KiCad isn't mostly fun. And it has taken a lot of time in the past, so I tried to do the work in the near past mainly for the major releases as the point releases are mostly non critical here. I already have given up to work on the documentation.

I'm probably the only person who has trying to do this in the past 2-3 years. But also in all these years I didn't get **one** single complain or contrary a remarkable contribution to kicad-i18n. And I started on kicad-i18n while about 85% of the strings are already translated.
I tried mostly to keep some translations in sync over the various places like Via, MicroVia and so on. But I'm also not a professional translator English-German and in particular for this specific technical language. And it did happen that some UI strings did change from A to B and to C but finally mostly again to A (with really small changes of the original A).

I personally not a big fan of mixing up a lot of English acronyms into the historical German language so I prefer "Durchkontaktierung" or "DoKu" over the original English word "Via". But as Simon wrote, this brings in some problems as the translated string isn't readable or truncated later in the UI. I bet German isn't the only language here with such problems.
Writing some compound nouns somehow not fully correct isn't a big issue for me, a lot of these words are for sure not in the German Duden so they officially do not really exist. For all other words which one might want to change my typical answer is (again), there is a mailing list to bring this to attention or at least write up some email to the people who have introduced these.

One more problem for me as someone who tries to translate the strings is I can't always follow the recent development which is sometime moving forward really quickly. In the end to quick for me. Plus some parts of the development are far away from my use cases.

As long as there is no team which is handling such and other problems by internal coordination I don't see much room for improvements. But I don't see team, even not in the near future. I wont have more time to work on l10n stuff for KiCad in the future.

I haven't a big and good proposal how to improve this all. Some things are obvious to me.

- Translations need to be done within a team to be effective and on high quality.
- Such teams need to met each other from time to time, often called a sprint.
- There need to be  codex and some basic rules how translations need to be handled.
- And there needs to be a feedback channel to the developers to know exactly what the UI is doing (or should do :smiley:  ).

GyrosGeier, 2019-04-18:
Absolutely. I've ignored it so far except for obvious problems, simply because I use KiCad in English most of the time.

Should we try to get a KiCad i18n-de session going some time soon? I could ask for some space at the Debian miniDebConf in Hamburg, or I could host something in Berlin.

tijuca, 2019-04-20:
> Should we try to get a KiCad i18n-de session going some time soon? I could ask for some space at the Debian miniDebConf in Hamburg, or I could host something in Berlin.

Feel free to drop a note to Holger, I'll be in Hamburg again this year. Last year I was first asking on the list for potential attendees but no interested people so far. I'm also from time to time in Berlin for a few days. I'm open for anything that will drive us forwards. As usual the main limitation is free time.

roybaer, 2019-04-23:
Well, I guess it all boils down to time, then.
I myself fixed the German translation of a late KiCAD 4 version back in 2017 and submitted that as a pull request (which got merged).
But when I tried that again in February, my translation became outdated before I could even submit a pull request and then I ran out of time.
It is hopelessly outdated, now, but maybe you want to quickly skim through it and let me know what you think w.r.t. correctness, style etc.

[de_work.patch.txt](https://github.com/KiCad/kicad-i18n/files/3107201/de_work.patch.txt)

Maybe there is a way to transfer that to a more recent version without doing everything by hand.

MichaelKavanagh, 2019-04-23:
Related issue from launchpad bug tracker: https://bugs.launchpad.net/kicad/+bug/1683813

### pcbnew: translation (DE) in Preset Layer Groupings

The translation to german of: "Four layers, parts on Front and Back" in Preset Layer Groupings is not correct. A correct translatation would be: "Vier Lagen, Bauteile auf Vorder- und Rückseite" (instead of "...nur auf Rückseite)

Application: kicad
Version: 4.0.6 release build
wxWidgets: Version 3.0.2 (debug,wchar_t,compiler with C++ ABI 1010,GCC 6.3.0,wx containers,compatible with 2.8)
Platform: Windows 7 (build 7601, Service Pack 1), 64-bit edition, 64 bit, Little endian, wxMSW
Boost version: 1.60.0
Curl version: libcurl/7.52.1 OpenSSL/1.0.2k zlib/1.2.11 libssh2/1.8.0 nghttp2/1.19.0 librtmp/2.3
         USE_WX_GRAPHICS_CONTEXT=OFF
         USE_WX_OVERLAY=OFF
         KICAD_SCRIPTING=ON
         KICAD_SCRIPTING_MODULES=ON
         KICAD_SCRIPTING_WXPYTHON=ON
         USE_FP_LIB_TABLE=HARD_CODED_ON
         BUILD_GITHUB_PLUGIN=ON

tijuca, 2019-04-24:
Not really, this report is about grammatical issues. The referenced bug report is around two years outdated.

> Version: 4.0.6 release build

herostrat, 2019-04-30:
I am interested in helping.

What are sensible ways to help housekeeping?

tijuca, 2019-05-01:
A good start would be to use the patch @roybaer has provided and try to add these changes to both branches (master and 5.1).

GyrosGeier, 2019-05-03:
FWIW, the patch applies on top of commit 0f23956f0, but rebasing that is still a hard problem.

GyrosGeier, 2019-05-03:
I've asked if we could get some space at MiniDebConf Hamburg (June 7th to 9th) for a meetup, if that is possible, who'd come (so I know if it makes sense)?

GyrosGeier, 2019-05-03:
Wait WTF I'm stupid. I don't have time that weekend. :/

tijuca, 2019-05-04:
I'll be in Hamburg. So if anybody will join feel free to come.

Rebasing PO files is a pain in in my eyes mostly impossible at some point. If necessary I try let the PO editor do the work. So apply the patch and load then the current source files. But i haven't had time to have a look.

herostrat, 2019-05-04:
I tried my best git-foo but I cannot get it to merge sensibly.
I guess it makes more sense to implement it by hand...

ciampix, 2019-05-04:
There is a trick though... rebasing is a problem due to clash not on strings but on comments. The thing is: you can manually get rid of comments, use the gettext toolset to merge the clashing two .po files (master and the file to rebase), and then update & merge. There should be a way to integrate this procedure locally to git in a form of script to create a non-conflicting merge and I found one once one on the Internet but I can't find it right now....

GyrosGeier, 2019-05-04:
The sort order also changed in between. `msgcat` can merge PO files, but alas not completely.

ciampix, 2019-05-04:
In fact thought about the msgmerge command of the xgettext toolset but I could be wrong, I am not an expert...

tijuca, 2019-05-06:
To prevent some doubled work, I started yesterday while traveling by train to adopt the patch from @roybaer into the recent 5.1 branch.

ciampix, 2019-06-06:
May I close this and move any discussion on the ml please?

tijuca, 2019-06-07:
In my eyes this can be closed.

ciampix, 2019-06-07:
Ok thanks!

closed 2019-06-07 by ciampix