KiCad/kicad-i18n #407: Random '&' characters in Polish translation

marecl, 2019-05-23:
For example (line 14244)
msgid "Hide &Grid"
msgstr "Ukryj siatkę"

(14228)
msgid "Print Pre&view..."
msgstr "Podgląd wydruku..."

Should they be included in translated string or can be removed?

francisco-pereira, 2019-05-23:
This character marks the "hot keys" on menu items (as an underline on windows), you should probably keep them on the translation.

marecl, 2019-05-23:
Probably yes. Is their position important or should they 'just be' there?

francisco-pereira, 2019-05-23:
The position is important as the next character following it is the "hot key".
I always try to use the same or most logical character.
As an example,  when translating "&Show Version Info"
I use "&Exibir Informações da Versão"
Here I don't have an "S" character in the translation, so I put the & in the word that has the same meaning as "Show" hence "&Exibir" marking the letter "E".
The only care i think you must take is to not mark the same character on two items of the same menu.

ciampix, 2019-06-06:
I use the same strategy for Italian... I'm closing this because it is not really an issue (and BTW this is not the right place to discuss this topic ... write to the docs/i18n ml instead...)

closed 2019-06-06 by ciampix