KiCad/kicad-i18n #446: Tag 5.1.4

nickoe, 2019-08-06:
On sunday I will tag the 5.1 branch for 5.1.4.

Please help check for untranslated or fuzzy strings before sunday if possible. Thank you.

nickoe, 2019-08-06:
Current status is:
de, en, fr, pt, ru, zh_CN are complete

Almost complete are:
it, pl, ja, hu and cs

nickoe, 2019-08-06:
@ciampix I see two fuzzy translations for it

nickoe, 2019-08-06:
@starfort-jp 
## ja
5594 translated messages
27 fuzzy translations
16 untranslated messages

nickoe, 2019-08-06:
@keruseykaryu as you contributed last time for the **pl** language I would like to poke you. Do you intend or want any updates for kicad **5.1.4**, there are:

5624 translated messages
12 fuzzy translations
1 untranslated message

nickoe, 2019-08-06:
@farkasgzoltan as you contributed last time for the **hu** language I would like to poke you. Do you intend or want any updates for kicad **5.1.4**, there are:

- 5546 translated messages
- 61 fuzzy translations
- 30 untranslated messages

nickoe, 2019-08-06:
@MichalJahelka as you contributed last time for the **cs** language I would like to poke you. Do you intend or want any updates for kicad **5.1.4**, there are:

- 5443 translated messages
- 130 fuzzy translations
- 64 untranslated messages

ciampix, 2019-08-07:
> @ciampix I see two fuzzy translations for it

Done, thanks!

keruseykaryu, 2019-08-07:
> @keruseykaryu as you contributed last time for the **pl** language I would like to poke you. Do you intend or want any updates for kicad **5.1.4**, there are:
> 
Done.

nickoe, 2019-08-07:
@MichalJahelka , @starfort-jp , @farkasgzoltan I will do a pull in the schedule and tag tonight, unless you reply before I tag  that you need more time.

nickoe, 2019-08-07:
Tagged.

closed 2019-08-07 by nickoe