KiCad/kicad-i18n #463: Pcbnew Wrong German Menue Translation

Jan-vi, 2019-10-07:
Pcbnew shows a menu item with Place->Dimension. For German translation, there are two diffrent expressions in use for "dimension". The German translation "Einfügen->Abmessungen" should change in "Einfügen->Bemaßung". For Austria and Switzerland "Bemassung" is ok (becouse of missing "scharfes Ess" in their character set) and will be also accepted for Germans. "Abmessungen" is also ok to understand after using once, but correct menue items makes the Kicad much more intuitive for beginners who consider to change from other EDA. "Abmessung" may be accepted for e.g. length=30mm if used for associative drawing input while "Bemaßung" describes the use of arrows graphics in all styles. For examples see the lesson here:

https://www.hs-anhalt.de/fileadmin/Dateien/FB6/personen/voigt_st/Lehrunterlagen/05_CAD/04_techZeich_Masseintragungen.pdf

Even Autocad is translating well:
https://knowledge.autodesk.com/de/support/autocad/getting-started/caas/CloudHelp/cloudhelp/2019/DEU/AutoCAD-Core/files/GUID-9810D35D-E975-49E2-88EA-51FE4ACC92DB-htm.html

Application: Pcbnew
Version: (5.1.4-87-g78af1daa2)-1, release build
Libraries:
wxWidgets 3.0.4
libcurl/7.61.1 OpenSSL/1.1.1 (WinSSL) zlib/1.2.11 brotli/1.0.6 libidn2/2.0.5 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.34.0
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
Boost: 1.68.0
OpenCASCADE Community Edition: 6.9.1
Curl: 7.61.1
Compiler: GCC 8.2.0 with C++ ABI 1013

Build settings:
USE_WX_GRAPHICS_CONTEXT=OFF
USE_WX_OVERLAY=OFF
KICAD_SCRIPTING=ON
KICAD_SCRIPTING_MODULES=ON
KICAD_SCRIPTING_PYTHON3=OFF
KICAD_SCRIPTING_WXPYTHON=ON
KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
KICAD_SCRIPTING_ACTION_MENU=ON
BUILD_GITHUB_PLUGIN=ON
KICAD_USE_OCE=ON
KICAD_USE_OCC=OFF
KICAD_SPICE=ON

poeschlr, 2019-10-07:
It would be news to me that austria does not have "ß".

chschlue, 2019-10-07:
It would be news to me that replacing ß with ss is acceptable in Germany :)

nickoe, 2019-10-07:
@Jan-vi would you propose a patch?

Jan-vi, 2019-10-15:
Swiss users may (who are probably many @ CERN) may choose between French and German itself. Here is my very first patch and I have no idea if that workflow is like intended:

[bug463patch.txt](https://github.com/KiCad/kicad-i18n/files/3731442/bug463patch.txt)

poeschlr, 2019-10-16:
I am not a translator so take this with a grain of salt:
It looks like this repo uses [pull requests](https://help.github.com/en/articles/about-pull-requests) to accept user contributions.

---

Which would mean you need to make a [fork](https://help.github.com/en/articles/fork-a-repo) of the repo, create a separate [branch](https://git-scm.com/book/en/v2/Git-Branching-Basic-Branching-and-Merging) and use that to create the pull request. You will then probably get feedback on your work which you can incorporate by pushing further commits to the branch used for your pull request.

Jan-vi, 2019-10-19:
While thinking about the reason on why to create a separate own branch instead working with the forked master clone, I found the latest commited de/po heavily out of synch with the kicad-source-mirror. There is more than one location to swap "Abmessung/Bemaßung" than in pulldown menu. Same and diffrent issues are inside the dimension property dialog. Trying xgetext to generate own pot file to msmerge is probably futile. Shell script cannot execute in windows (without msys) while the po file takes account for all kicad source files. "Line-Thickness" of dimension property dialog doesnt appear in the po file at and is probably not automatically detected becouse of forgotten tool-tip text in the sources. Dialog_text_properties.cpp is used for more than only the dimension arrow setup and generated by wxFormBuilder. Latest wxFormBuilder shows incompability warnings while converting the kicad-source-mirror fbp file. Seems I have to to pass the next black belt examination first and thereby practice with more simple problems ...

nickoe, 2019-10-19:
@Jan-vi if you like, I can update the po file against the source and then you can update the translation on top of that.

Jan-vi, 2019-10-19:
thanks, @nickoe, I can try to do my best

nickoe, 2019-10-19:
@Jan-vi I updated the po file for de on both master and the 5.1 branch. Please make your pull request on top of those changes.

nickoe, 2019-11-15:
Merged in Merged in https://github.com/KiCad/kicad-i18n/commit/b489f707305e83d30c3c268d3fb6587bfff65e25 for the 5.1 branch

commit b489f707 (2019-11-15): Update German to 5.1 PCB Dimension Dialogs

Fixes #463
closed 2019-11-15 by nickoe