KiCad/kicad-i18n #473: Tag 5.1.5

nickoe, 2019-11-15:
On sunday I will tag the 5.1 branch for 5.1.5.

Please help check for untranslated or fuzzy strings before sunday (17th of november) if possible. Thank you.

I know it is a short notice, but the rc has been announced some time ago and hopefully only minimal changes are needed.

nickoe, 2019-11-15:
Current status is:
en, fr, it, pt, ru, zh_CN are complete

Almost complete are:
de, pl, ja, hu and cs

nickoe, 2019-11-15:
@keruseykaryu as you contributed last time for the pl language I would like to poke you. Do you intend or want any updates for kicad 5.1.5, there are:

- 5626 translated messages
- 22 fuzzy translations
- 5 untranslated messages

nickoe, 2019-11-15:
@tijuca  as you contributed recently for the **de** language I would like to poke you. Do you intend or want any updates for kicad 5.1.5, there are:

- 5627 translated messages
- 21 fuzzy translations
- 5 untranslated messages

tijuca, 2019-11-16:
O.k. done it right now (#474), otherwise I'm impossible to do any thing due time constrains.

keruseykaryu, 2019-11-16:
I am _out of business_. Sorry.
If anyone else (Daniel, Marek...) is willing to improve it, please do so.

nickoe, 2019-11-16:
@tijuca cool, thank you.

closed 2019-11-17 by nickoe